AtsushiSakai/PythonRobotics #25: dynamic window maximum recursion depth exceeded

violetguos, 2018-01-16:
Hi, I just cloned your repository today and I have a problem running ```dynamic_window_approach.py```with this version's [```matplotlibrecorder```](https://github.com/AtsushiSakai/matplotrecorder/tree/adb95ae92b0b7384fed7c1b39dd41e3ab86bad78) 
The error message is shown below
```
Traceback (most recent call last):
  File "dynamic_window_approach.py", line 219, in <module>
    main()
  File "dynamic_window_approach.py", line 199, in main
    plt.plot(ob[:, 0], ob[:, 1], "ok")
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/pyplot.py", line 2987, in plot
    ret = ax.plot(*args, **kwargs)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/axes.py", line 4137, in plot
    for line in self._get_lines(*args, **kwargs):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/axes.py", line 317, in _grab_next_args
    for seg in self._plot_args(remaining, kwargs):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/axes.py", line 295, in _plot_args
    x, y = self._xy_from_xy(x, y)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/axes.py", line 213, in _xy_from_xy
    bx = self.axes.xaxis.update_units(x)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/axis.py", line 1336, in update_units
    converter = munits.registry.get_converter(data)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/units.py", line 148, in get_converter
    converter = self.get_converter(xravel[0])
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/units.py", line 148, in get_converter
    converter = self.get_converter(xravel[0])
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/units.py", line 148, in get_converter
    converter = self.get_converter(xravel[0])
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/units.py", line 148, in get_converter
    converter = self.get_converter(xravel[0])
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/units.py", line 148, in get_converter
    converter = self.get_converter(xravel[0])
#keeps repeating line 148 error until
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/numpy/matrixlib/defmatrix.py", line 316, in __getitem__
    out = N.ndarray.__getitem__(self, index)
RuntimeError: maximum recursion depth exceeded
```

AtsushiSakai, 2018-01-16:
@violetguos Hi. Thank you for your contribution. 
Actually, matplotlibrecorder is not needed anymore. So I remove it.
Could you update the code and try it?

violetguos, 2018-01-17:
I updated and now I get a different maximum recursion exceeded error
```
Traceback (most recent call last):
  File "dynamic_window_approach.py", line 218, in <module>
    main()
  File "dynamic_window_approach.py", line 200, in main
    plt.plot(ob[:, 0], ob[:, 1], "ok")
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/pyplot.py", line 2987, in plot
    ret = ax.plot(*args, **kwargs)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/axes.py", line 4137, in plot
    for line in self._get_lines(*args, **kwargs):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/axes.py", line 317, in _grab_next_args
    for seg in self._plot_args(remaining, kwargs):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/axes.py", line 295, in _plot_args
    x, y = self._xy_from_xy(x, y)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/axes.py", line 213, in _xy_from_xy
    bx = self.axes.xaxis.update_units(x)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/axis.py", line 1336, in update_units
    converter = munits.registry.get_converter(data)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/units.py", line 148, in get_converter
    converter = self.get_converter(xravel[0])
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/units.py", line 148, in get_converter
    converter = self.get_converter(xravel[0])

######Repeat the line 148 error until maximum recursion depth exceeded########

  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/numpy/matrixlib/defmatrix.py", line 316, in __getitem__
    out = N.ndarray.__getitem__(self, index)
RuntimeError: maximum recursion depth exceeded
```
I also get this error when I try to run ```\test\test_dynamic_window_approach.py  ```

```
Traceback (most recent call last):
  File "test_dynamic_window_approach.py", line 3, in <module>
    from PathPlanning.DynamicWindowApproach import dynamic_window_approach as m
ImportError: No module named PathPlanning.DynamicWindowApproach
```

AtsushiSakai, 2018-01-17:
Could you please let me know your versions of python and matplotlib.
In my env, it doesn't happen..

violetguos, 2018-01-17:
Sorry I was running it in Python 2.7. It is now working when compiled in Python 3.6. Thank you for your reply!

AtsushiSakai, 2018-01-17:
@violetguos OK. Thanks.

closed 2018-01-17 by AtsushiSakai