AtsushiSakai/PythonRobotics #41: KeyError in Mix Integer Path Planning

carlosluisg, 2018-03-05:
Hi, I'm interested in modifying your Mix Integer Path Planning to a multi-vehicle scenario, however I'm having troubles running your code. This is the console output when I run the code:

mix_integer_opt_path_planning.py start!!
('time:', 0)
Academic license - for non-commercial use only
Traceback (most recent call last):
  File "mix_integer_opt_path_planning.py", line 127, in <module>
    main()
  File "mix_integer_opt_path_planning.py", line 101, in main
    s_p, u_p = control(s, gs, ob)
  File "mix_integer_opt_path_planning.py", line 69, in control
    prob.solve(solver=cvxpy.GUROBI)
  File "/usr/local/lib/python2.7/dist-packages/cvxpy-0.4.11-py2.7.egg/cvxpy/problems/problem.py", line 209, in solve
    return self._solve(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/cvxpy-0.4.11-py2.7.egg/cvxpy/problems/problem.py", line 331, in _solve
    kwargs)
  File "/usr/local/lib/python2.7/dist-packages/cvxpy-0.4.11-py2.7.egg/cvxpy/problems/solvers/gurobi_intf.py", line 229, in solve
    A, b)
  File "/usr/local/lib/python2.7/dist-packages/cvxpy-0.4.11-py2.7.egg/cvxpy/problems/solvers/gurobi_intf.py", line 326, in add_model_lin_constr
    expr_list[i].append((c, v))
KeyError: 206

It has something to do with the use of the GUROBI solver, however I verified it is correctly installed already (I ran one of the examples they provide). Do you have an idea of what could be wrong? Thanks!

AtsushiSakai, 2018-03-05:
Hi. @carlosluisg. Thank you for your interest about my repo.
I have no idea about it actually... However, you should use Python 3.6. It seems you are using 2.7.
I hope your problem will be solved if you upgrade your python.

carlosluisg, 2018-03-05:
I tried that, installing cvxpy in Python 3. However I'm getting an error saying cvxpy has no attribute Bool now

AtsushiSakai, 2018-03-05:
@carlosluisg Could you paste the new error msg?

carlosluisg, 2018-03-05:
Sure, here it is:

mix_integer_opt_path_planning.py start!!
time: 0
Traceback (most recent call last):
  File "mix_integer_opt_path_planning.py", line 127, in <module>
    main()
  File "mix_integer_opt_path_planning.py", line 101, in main
    s_p, u_p = control(s, gs, ob)
  File "mix_integer_opt_path_planning.py", line 39, in control
    o = cvxpy.Bool(4 * nob, T)
AttributeError: module 'cvxpy' has no attribute 'Bool'

Once again, thanks for the support

AtsushiSakai, 2018-03-06:
Which versions of cvxpy are you using?

carlosluisg, 2018-03-06:
1.0.0

AtsushiSakai, 2018-03-06:
Oh, I'm using 0.4. Can you downgrade it? or it takes time to fix the problem..

carlosluisg, 2018-03-06:
I downgraded but now the problem seems to be that grunobi does not support Python 3.5 in Linux, what is your current OS?

AtsushiSakai, 2018-03-06:
I'm using Mac (High sierra). Why you are using Python 3.5? I think cvxpy 0.4 supports Python 3.6.

carlosluisg, 2018-03-06:
Yes, I just realized that. I'm upgrading now and will get back to you with the results. Hopefully it will work

carlosluisg, 2018-03-06:
The problem Persists with Python3.6 and cvxpy 0.4. Output as proof:

time: 0
Academic license - for non-commercial use only
Traceback (most recent call last):
  File "mix_integer_opt_path_planning.py", line 127, in <module>
    main()
  File "mix_integer_opt_path_planning.py", line 101, in main
    s_p, u_p = control(s, gs, ob)
  File "mix_integer_opt_path_planning.py", line 69, in control
    prob.solve(solver=cvxpy.GUROBI)
  File "/usr/local/lib/python3.6/dist-packages/cvxpy-0.4.11-py3.6.egg/cvxpy/problems/problem.py", line 209, in solve
    return self._solve(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/cvxpy-0.4.11-py3.6.egg/cvxpy/problems/problem.py", line 331, in _solve
    kwargs)
  File "/usr/local/lib/python3.6/dist-packages/cvxpy-0.4.11-py3.6.egg/cvxpy/problems/solvers/gurobi_intf.py", line 229, in solve
    A, b)
  File "/usr/local/lib/python3.6/dist-packages/cvxpy-0.4.11-py3.6.egg/cvxpy/problems/solvers/gurobi_intf.py", line 326, in add_model_lin_constr
    expr_list[i].append((c, v))
KeyError: 240

I'm running Ubuntu 16.04, maybe there's a bug linux related, since it works for you on OSX

AtsushiSakai, 2018-03-06:
I'm using gurobipy-7.5.2... It's same.

closed 2018-03-08 by AtsushiSakai