AtsushiSakai/PythonRobotics #46: Cubic spline planning: Spline2D doesn't generate spline from closely spaced x-y waypoints

babylonman, 2018-03-21:
for example: 
x-y waypoints which are closely spaced do not show a spline when run with main(), the Spline 2D test 
x = [random.uniform(1, 1.05) for i in range(5)]
y = [random.uniform(1, 1.05) for i in range(5)]

while larger values do not show this as an issue
x = [random.uniform(1, 11.05) for i in range(5)]
y = [random.uniform(1, 11.05) for i in range(5)]

This would be useful for GPS lat long coordinates which have small absolute differences corresponding to reasonably large real life distances.
x = [37.3779926, 37.3778115, 37.3777806, 37.3779436]
y = [-122.0419519, -122.0420216, -122.0417949, -122.0417266]

AtsushiSakai, 2018-03-21:
Hi @babylonman . Thank you for your interest.
Can you post your actual code here?.
I will test it.

babylonman, 2018-03-21:
@AtsushiSakai , I've attached an example based on the original test you included in cubic_spline_planner.py. 

Thanks for the great collection of resources by the way!

[cubic_spline_planner_example.py.zip](https://github.com/AtsushiSakai/PythonRobotics/files/1833530/cubic_spline_planner_example.py.zip)

AtsushiSakai, 2018-03-21:
@babylonman Thank you for sending your code.
There is a solution about your problem.
You should make the third input of np.arange smaller, because this input means the sampling distance of the interpolated path.
Your x-y points are sampled in a very narrow area, this means you have to choose smaller value for it.

Like:
   s = np.arange(0, sp.s[-1], 0.001)

If I set the value, I can get this result with your code. It seems good.

[image]
[image]

closed 2018-03-22 by babylonman