AtsushiSakai/PythonRobotics #54: Hybrid A* planner simulation should be added

AtsushiSakai, 2018-03-29:


jwdinius, 2018-09-17:
I'll have a crack at this in the next couple of weeks.  I have implemented in the past so it shouldn't be too difficult.  Do you still want this @AtsushiSakai?

AtsushiSakai, 2018-09-17:
@jwdinius Hi. Thank you for your suggestion. Yes. I still want it!!. Your PR is welcome.

xinbada007, 2018-11-14:
Hey What's the progress now? Anyone adding the Hybrid-A* simulation? 

Just wondering if it just misses the simulation implementation but the algorithm is complete?

AtsushiSakai, 2018-11-15:
@xinbada007 I'm waiting a someone's PR.

jwdinius, 2018-11-16:
I have been unable to complete this work yet.  I will try to have a PR by Christmas @xinbada007 @AtsushiSakai

AtsushiSakai, 2018-11-16:
@jwdinius Thank you. Take your time : ) .

renegadephysicist, 2018-11-25:
There is an implementation [here in Julia ](https://github.com/XiaojingGeorgeZhang/OBCA/blob/master/AutonomousParking/hybrid_a_star.jl) that might be useful.

jwdinius, 2018-12-29:
I haven't forgotten about this.  I missed the Christmas timeline, but I will be working on it this weekend @AtsushiSakai @xinbada007

AtsushiSakai, 2018-12-30:
@jwdinius Thank you for your comment. Take your time and enjoy the end of the year.

AtsushiSakai, 2019-03-03:
@jwdinius Hi. Are you working this issue?

jwdinius, 2019-03-03:
I'm sorry @AtsushiSakai, I've been unable to find the time to finish this.  If anyone has the time before I can finish, here's a repo I am translating to python https://github.com/informramiz/Hybrid-Astar-search-algorithm.git.  I will continue working on it, but I have no estimated time of completion.

zhengzh, 2019-03-04:
BFS algorithm in Informaramiz's repo is wrong actually.

jwdinius, 2019-03-06:
@zhengzh thanks for the response.  The implementation generates reasonable looking paths.  What specifically is wrong about the implementation?  If you know of something else to use as a starting point, I would be interested to hear about it.

zhengzh, 2019-03-07:
@jwdinius , Hi. As we know, in BFS algorithm, we should put every children of a parent node into the queue, but he only puts the minimum cost node into the queue (in [line](https://github.com/informramiz/Hybrid-Astar-search-algorithm/blob/419e181e7eec982d25f710d0610d2214e7d0047e/HybridAstarAlgorithm/src/hybrid_breadth_first.cpp#L558)), but forgets to put other nodes into the queue. I test the program with the `MAZE` (not the `EMPTY_MAZE`) and it gives 60 steps to reach the goal, but in this [repo](https://github.com/ChenBohan/Robotics-Path-Planning-03-Hybrid-A-Star), it uses A* algorithm (not hybrid A*) and only needs 38 steps to reach the goal, and we know both algorithms should give the same optimal path. I don't know why the file is named as `hybrid_breadth_first.cpp`, I think BFS won't help save expanded nodes. The best and simplest hybrid A* implementation I have found is in this repo https://github.com/AtsushiSakai/HybridAStarTrailer, which is the just the same as the original [paper](https://ai.stanford.edu/~ddolgov/papers/dolgov_gpp_stair08.pdf) described, although I never learn julia language before, I can still understand the code :smile_cat:.

jwdinius, 2019-03-10:
Ok @zhengzh , your points are well-taken.  I have looked at the implementation and I will work to translate it to python 3.6.  Do we want a more general hybrid a* implementation (like the A* implementation)?  I will do the translation so that the core search algorithm can be easily imported into possible future hybrid a* implementations.

zhengzh, 2019-03-13:
Nice! @jwdinius, Thank you, please make the `MAZE` a little bit harder like driving a car into a parking space.
[image]

jwdinius, 2019-03-24:
https://github.com/AtsushiSakai/PythonRobotics/pull/180 addresses this, so I'm stopping work

AtsushiSakai, 2019-04-03:
@jwdinius @zhengzh I merged #180 . Thank you so much for your help!!

closed 2019-04-03 by AtsushiSakai