AtsushiSakai/PythonRobotics #58: difference between PythonRobotics and MATLABRobotics

JSRGFJZ, 2018-05-05:
Hi, I am reading probabilistic robotics this book, just find this repository, I have some questions:
1. is there any difference between PythonRobotics and MATLABRobotics?
2. all code corresponding to the <<probabilistic robotics>>
3. any toturials?

AtsushiSakai, 2018-05-06:
Hi @JSRGFJZ 
Thank you for your interest to my repo.

1.
MATLABRobotics is my old project, and it is no longer active.

2.
I'm sorry I cannot understand your question.. 
Could you add the details of the question?

3.
I'm sorry there is no tutorial.
However only thing you have to do is just run the python script on each directory.
You can see its simulation.

Best

JSRGFJZ, 2018-05-06:
@AtsushiSakai for question2 , I mean all the code are written following the book  probability robotics? actually I am wondering if I can use this code to grasp the key idea of this book probabilistic book?

AtsushiSakai, 2018-05-06:
@JSRGFJZ OK. I see.
Some of codes are based on Probabilistic robotics, for example EKF, UKF, PF, EKFSLAM, FastSLAM.
However, others are not like path planning algorithms and path following one.

closed 2018-05-11 by AtsushiSakai