AtsushiSakai/PythonRobotics #61: [Propose] Path planning geometric method

tsaoyu, 2018-05-30:
First, I would like to say thank you to the author(s) and the contributor(s). This project is really awesome! Visualisation and explanation are just to the sweet point. 

Of course, I want to make some contribution to this project. Does it make sense to include some other geometric based method in path planning section? i.e. visibility graph method

Many thanks,

AtsushiSakai, 2018-05-31:
@tsaoyu Hi. Thank you for your interest about this repo. I'm so sorry that A small PR is welcome, but the new algorithm code is not. Because this is my personal project for learning myself.

tsaoyu, 2018-05-31:
@AtsushiSakai sure, your argument makes sense to me.

closed 2018-05-31 by tsaoyu