AtsushiSakai/PythonRobotics #65: something was wrong with model_predictive_speed_and_steer_control.py

chouer19, 2018-06-05:
model_predictive_speed_and_steer_control.py start!!
Traceback (most recent call last):
  File "model_predictive_speed_and_steer_control.py", line 543, in <module>
    main()
  File "model_predictive_speed_and_steer_control.py", line 520, in main
    t, x, y, yaw, v, d, a = do_simulation(cx, cy, cyaw, ck, sp, dl)
  File "model_predictive_speed_and_steer_control.py", line 403, in do_simulation
    xref, x0, dref, oa, odelta)
  File "model_predictive_speed_and_steer_control.py", line 235, in iterative_linear_mpc_control
    oa, od, ox, oy, oyaw, ov = linear_mpc_control(xref, xbar, x0, dref)
  File "model_predictive_speed_and_steer_control.py", line 262, in linear_mpc_control
    cost += cvxpy.quad_form(u[:, t], R)
  File "/usr/local/lib/python3.5/dist-packages/cvxpy/expressions/expression.py", line 313, in __getitem__
    return cvxtypes.index()(self, key)
  File "/usr/local/lib/python3.5/dist-packages/cvxpy/atoms/affine/index.py", line 48, in __init__
    self.key = ku.validate_key(key, expr.shape)
  File "/usr/local/lib/python3.5/dist-packages/cvxpy/utilities/key_utils.py", line 49, in validate_key
    raise IndexError("Too many indices for expression.")
IndexError: Too many indices for expression.

chouer19, 2018-06-05:
the version of cvxpy should be 0.4......

AtsushiSakai, 2018-06-08:
@chouer19 I'm sorry, I am still using v0.4 cvxpy mainly. 
If you want to use v1.0 or higher , PR is wellcome!!

closed 2018-06-08 by AtsushiSakai