AtsushiSakai/PythonRobotics #67: PathPlanning algorithms(A* and Dijkstra) not showing final paths

kartikmadhira1, 2018-06-13:
**System Information -**
Python version - 3.6.5
Matplotlib version - 2.2.2
System OS - MacOS High Sierra 

**Description**
Running the Dijkstra python file, runs the grid search perfectly, but as soon as the goal is reached, the plot hangs and is unable to plot the final path. 
Is this a library issue? I modified it for my system by plotting the path in the final path calculation function:

def calc_final_path(ngoal, closedset, reso):
    # generate final course
    rx, ry = [ngoal.x * reso], [ngoal.y * reso]
    pind = ngoal.pind
    while pind != -1:
        n = closedset[pind]
        rx.append(n.x * reso)
        ry.append(n.y * reso)
        plt.plot(rx, ry, "xr")
        plt.pause(0.000001)
        pind = n.pind

    return rx, ry

AtsushiSakai, 2018-06-14:
@kartikmadhira1 Did the original code work correctly before you modified it? and does other script work?

kartikmadhira1, 2018-06-14:
The original code was working but only till the goal was reached and then the final path isn't traced. I just plotted the final path (final path calculation function ) itself so now it traces the path.

See this picture for reference(error):
[image]

AtsushiSakai, 2018-06-16:
@kartikmadhira1 I'm sorry, I don't know what is happening. Maybe that is a problem from matplotlib or other library. I cannot help you. sorry.

closed 2018-06-16 by AtsushiSakai