AtsushiSakai/PythonRobotics #70: Documentation for bezier and stanley control ?

araffin, 2018-06-18:
Hello,
First, i would like to congratulate you for this great repo =).

Then, I would like to know if you would be interested with a PR adding documentation to bezier curve and stanley control.
In fact, they are now used on a real robot:
https://github.com/sergionr2/RacingRobot/blob/master/path_planning/bezier_curve.py
https://github.com/sergionr2/RacingRobot/blob/master/path_planning/stanley_controller.py

I added support for computing bezier curve with n control points along with derivative and curvature computation. The bezier curve can thus be used with the stanley controller now.

before i submit a PR, do you have any style convention for comments/code ?

AtsushiSakai, 2018-06-19:
@araffin Hi. Thank you for your interest. If you create a PR for this repo. Please see PEP 257 -- Docstring Conventions | Python.org https://www.python.org/dev/peps/pep-0257/ for comments and PEP 8 -- Style Guide for Python Code | Python.org https://www.python.org/dev/peps/pep-0008/ for code style. Best.

araffin, 2018-06-19:
Ok, thanks, i'll do that when i got some time ;)

closed 2018-06-19 by araffin