AtsushiSakai/PythonRobotics #77: Documentation using Jupyter notebook

AtsushiSakai, 2018-07-04:
.ipynb file may be good for documentation of each algorithm.

node-py, 2018-07-10:
I can help with jupyter notebook for each algorithm.

kartikmadhira1, 2018-07-13:
I have too implemented similar codes taking help from some of the codes in the Repo so I can help too. Can you be more specific regarding the documentation? As in to add in-depth text (math, diagrams, etc.) to each block of code or just superficially explain as to what each block of code does?

AtsushiSakai, 2018-07-13:
@kartikmadhira1 Thank you for your comment. Yes, it is great to include it's math, algorithm, and diagram. I think code description is not needed because there are comments on the code. If you read the doc and code, you can understand the basic idea of the algorithm. This is the goal of this project.

kartikmadhira1, 2018-07-13:
@AtsushiSakai Sure then. I'll try to get a first draft version on Astar and Dijkstra and present it to you. Will try to get it done by end of next week.

AtsushiSakai, 2018-07-13:
@kartikmadhira1 Oh. great. I appreciate your help!!.

AtsushiSakai, 2018-07-16:
I'm trying to write a jupyter notebook for MPC.
[PythonRobotics/notebook\.ipynb at master · AtsushiSakai/PythonRobotics](https://github.com/AtsushiSakai/PythonRobotics/blob/master/PathTracking/model_predictive_speed_and_steer_control/notebook.ipynb)

- [ ] finish jupyter notebook for MPC

kartikmadhira1, 2018-07-17:
@AtsushiSakai I am working on writing documentation regarding [Kalman Filters](https://github.com/kartikmadhira1/Robotics-Algorithms/blob/master/Filters/KF_notes.ipynb)- concepts and math with examples from a few sources. We can add it to the EKF section if you want.

AtsushiSakai, 2018-07-17:
@kartikmadhira1 Nice. Could you make PR?

AtsushiSakai, 2018-07-21:
MPC's notebook is done. Next one is EKF.

- [ ] [PythonRobotics/notebook\.ipynb at master · AtsushiSakai/PythonRobotics](https://github.com/AtsushiSakai/PythonRobotics/blob/master/Localization/extended_kalman_filter/notebook.ipynb)

nithin1995, 2018-07-24:
Hi, I am interested to contribute. I am new to this field, and interested to learn and contribute. I would like you suggest an algorithm for me to start working on the documentation.

AtsushiSakai, 2018-07-25:
@nithin1995 Thank you. How about EKF? [PythonRobotics/notebook\.ipynb at master · AtsushiSakai/PythonRobotics](https://github.com/AtsushiSakai/PythonRobotics/blob/master/Localization/extended_kalman_filter/notebook.ipynb)

sunnytokyo, 2018-07-27:
hi atushi, I wud also like to contribute for codes and docs. Do contributors get to feature as authors for the project?

daniel-s-ingram, 2018-09-23:
I'll start working on Jupyter notebooks for the algorithms I've submitted!

AtsushiSakai, 2018-09-24:
@daniel-s-ingram That's great!! Your PR is welcome.

AtsushiSakai, 2018-11-19:
#130

adelizer, 2018-12-15:
hi @AtsushiSakai, I would like to help with creating a documentation notebook for fastSLAM1.0 if it is not in progress yet.

amirx96, 2019-03-26:
@AtsushiSakai Is there any documentation for the LQR steer controller? The closest source I could find was https://arxiv.org/pdf/1803.03758.pdf on page 10.

I am not sure how you incorporate dt into the A matrix.

AtsushiSakai, 2019-04-04:
I'm closing this issue once for v1.0 release.

closed 2019-04-04 by AtsushiSakai