AtsushiSakai/PythonRobotics #86: MPC have some problem

hexiangquan, 2018-07-14:
when i plan a path along x positive direction means zero degree.if current car's degree is 359 or 0 degree.that will cause problem .yaw error may be -359 or -1

hexiangquan, 2018-07-15:
I will share with you my code and my simulator . You code use model not real or not match real car’s model  .so car’s angle will not jitter

hexiangquan, 2018-07-15:
You can  image that  mpc control will make Xref -X →0 . Rlt=Xref[3:t]-x[3:t] when path alongside Zero direction and real car’s yaw maybe jitter between 1 degree and 359 degree .Rlt will be negative whether 1 degree or 359 . That cause problem

AtsushiSakai, 2018-07-16:
@hexiangquan I'm not sure that will happen, because yaw angle in MPC is updated by integral. I am not using pi_2_pi function inside MPC function. So, the angle will be smooth. If the yaw angle of reference trajectory is also smooth, the problem will no happen.

hexiangquan, 2018-07-16:
line 382 set  state = State(x=0, y=0, yaw=-math.pi, v=0.0)
get_straight_course  set like below code   you will see the phenomenon
def get_straight_course(dl):
    # ax = [0.0, 5.0, 10.0, 20.0, 30.0, 40.0, 50.0]
    # ay = [0.0, -1.0, 1.0, 0.0, -1.0, 1.0, 0.0]
    ax = [0.0, -10.0, -20.0, -40.0, -50.0, -60.0, -70.0]
    ay = [0.0, 1.0, -1.0, 0.0, -1.0, 1.0, 0.0]

    cx, cy, cyaw, ck, s = cubic_spline_planner.calc_spline_course(
        ax, ay, ds=dl)

    return cx, cy, cyaw, ck

traverse over math.pi line will cause problem

hexiangquan, 2018-07-16:
def get_straight_course(dl):
    # ax = [0.0, 5.0, 10.0, 20.0, 30.0, 40.0, 50.0]
    # ay = [0.0, -1.0, 1.0, 0.0, -1.0, 1.0, 0.0]
    ax = [0.0, -10.0, -20.0, -40.0, -50.0, -60.0, -70.0]
    ay = [0.0, 1.0, -1.0, 0.0, -1.0, 1.0, 0.0]

    cx, cy, cyaw, ck, s = cubic_spline_planner.calc_spline_course(
        ax, ay, ds=dl)

    return cx, cy, cyaw, ck

hexiangquan, 2018-07-16:
line297 
  +print(x0)
  +print(xref[3, :])
you will see which cause problem

hexiangquan, 2018-07-16:
if you set   state = State(x=0, y=0, yaw=math.pi, v=0.0)  car's init state 
get_straight_course set like below ,you will also encounter this problem
def get_straight_course(dl):
    ax = [0.0, -10.0, -20.0, -40.0, -50.0, -60.0, -70.0]
    ay = [0.0, -1.0, 1.0, 0.0, -1.0, 1.0, 0.0]

    cx, cy, cyaw, ck, s = cubic_spline_planner.calc_spline_course(
        ax, ay, ds=dl)

    return cx, cy, cyaw, ck

AtsushiSakai, 2018-07-16:
@hexiangquan I see. You mean if the initial yaw is not smooth to the reference trajectory, the problem will happen. I added initial yaw compensation code. Could you check if you problem solved or not?

hexiangquan, 2018-07-16:
When car is in back mode .there also have some problem

AtsushiSakai, 2018-07-16:
@hexiangquan In my test case, it works. Could you let me know your test code?

hexiangquan, 2018-07-16:
In back mode . Steer angle should multiply -1

AtsushiSakai, 2018-07-16:
@hexiangquan I'm sorry, I cannot your comment.

closed 2018-07-16 by AtsushiSakai

hexiangquan, 2018-07-16:
Sorry you test code cant test this situation. I have a simulator .i will share with you later