AtsushiSakai/PythonRobotics #87: A little errata?

soblin, 2018-07-15:
I think the following line in PathTracking/model_predictive_speed_and_steer_control.py, line 91, 
`A[3, 2] = DT * math.tan(delta)`

should be

`A[3, 2] = DT * math.tan(delta) / WB`

instead, because this term is the derivative of 

`phi_next = phi + DT * math.tan(delta) * v / WB`

with respcect to v. I tested the modified code, to have no problem.

BTW, could you teach me  the meaning of matrix C in ```get_linear_model_matrix```? I know the matrix A and B is the jacobian with respect to x and u, but what does C stand for?

AtsushiSakai, 2018-07-16:
@soblin Thank you for bug report. Yes, you are right. I fixed it. Thank you.

>what does C stand for?
This is a constant term of linearized model. See This reference: MPC Course Material - MPC Lab @ UC-Berkeley http://www.mpc.berkeley.edu/mpc-course-material 
I'm trying to write a document about it later.

closed 2018-07-16 by AtsushiSakai

AtsushiSakai, 2018-07-17:
@soblin Please check it.

https://github.com/AtsushiSakai/PythonRobotics/blob/master/PathTracking/model_predictive_speed_and_steer_control/notebook.ipynb