AtsushiSakai/PythonRobotics #93: Summary paper

AtsushiSakai, 2018-08-04:
I'm writing a summary paper of this repo to introduce it for academic people.
I'm planning to submit it to arXiv.

- [AtsushiSakai/PythonRoboticsPaper: PythonRoboticsPaper](https://github.com/AtsushiSakai/PythonRoboticsPaper)

AtsushiSakai, 2018-08-11:
TODO

- First version release
- Check 1
- Check 2
- Check 3
- Author management

AtsushiSakai, 2018-08-12:
@daniel-s-ingram @jwdinius @karanchawla @araffin @AlexisTM 
Hi all contributors. I appreciate so mush your help.

I am writing a summary paper about this repo to introduce it for more many robotics lovers.
I'm planning to submit it to arXiv. (maybe after the number of stars reached over 3000 stars)
This is the first draft of the paper. (not finished)
- [PythonRoboticsPaper/PythonRoboticsPaper\.pdf at master · AtsushiSakai/PythonRoboticsPaper](https://github.com/AtsushiSakai/PythonRoboticsPaper/blob/master/PythonRoboticsPaper.pdf)

I'm welcome to add your name as a co-author.
If you are interested in, please let me know your print name, current affiliation, e-mail address or your HP url.

Best.

AlexisTM, 2018-08-13:
Dear Atsushi, I would love it;

Alexis Paques,
Lead Robotics Engineer at Unmanned Life
Alexis.paques@gmail.com or Alexis@unmanned.life

karanchawla, 2018-08-13:
Hi Atsushi,

I'd be delighted as well.  However, do let me know how I can help. I can review it as well as contribute to it :) 

Here are my details: 
Karan Chawla,
Autonomy Software Engineer at Skyryse Inc. 
karanchawla0709@gmail.com or karan@skyryse.com

araffin, 2018-08-13:
Hi @atsushi-sakai ,

Thanks for the proposition. Please let us know if you want some feedback on the paper ;)

My details:
Antonin Raffin,
Research Engineer at ENSTA ParisTech
antonin.raffin@ensta-paristech.fr

daniel-s-ingram, 2018-08-13:
Hi @AtsushiSakai,

Wow, thank you for the offer! I would also be happy to help - please let me know if there is anything I can do :)

jwdinius, 2018-08-13:
@AtsushiSakai 
Thank you for the acknowledgment.  Please let me know what you would like from me to best assist you.
My details are:
Joseph Dinius
Staff Reseach Scientist - Controls & Perception at inVia Robotics, Inc.
jdinius@inviarobotics.com

daniel-s-ingram, 2018-08-13:
@AtsushiSakai,

Would you be interested in adding some robotic arm algorithms to the repo?

AtsushiSakai, 2018-08-14:
@daniel-s-ingram @jwdinius @karanchawla @araffin @AlexisTM
Thanks guys! I added you as authors:
[PythonRoboticsPaper/PythonRoboticsPaper\.pdf at master · AtsushiSakai/PythonRoboticsPaper](https://github.com/AtsushiSakai/PythonRoboticsPaper/blob/master/PythonRoboticsPaper.pdf)

If the paper is ready to review, I will let you guys know.
It must be helpful because I am not native English speaker.

AtsushiSakai, 2018-08-14:
@daniel-s-ingram Oh. Great. It must be helpful for robot lovers. If the arm simulation has a cool animation, it will be best!!. I'm welcome your PR.

AtsushiSakai, 2018-08-18:
@daniel-s-ingram @jwdinius @karanchawla @araffin @AlexisTM
Hi everyone.
I uploaded the first draft of the paper.
- [AtsushiSakai/PythonRoboticsPaper: PythonRoboticsPaper](https://github.com/AtsushiSakai/PythonRoboticsPaper)

If you have time, please check the draft and give me any feedbacks.
Feel free to add an issue and a PR for the paper repo. (it includes tex files)

Best

jwdinius, 2018-08-18:
It might be best to break the paper up into sections, I think.  This way each person could have some sections that they are responsible for.  What do you think @AtsushiSakai @AlexisTM @karanchawla @araffin @daniel-s-ingram ?

araffin, 2018-08-18:
I don't know if I have more time to dedicate than just giving feedback (I am currently quite busy at writing other articles and blog posts :/ ). What do you mean exactly by "responsible" ? in charge of writing a given section ?

jwdinius, 2018-08-18:
@araffin I only mean a section to review and make comments on.  The paper seems to be in a review stage.  This way we wouldn't have to review the whole paper but could provide feedback and suggestions on a few sections only.

AtsushiSakai, 2018-08-18:
@jwdinius Thank you for your suggestion. Only reviewing a few sections which you are interesting is OK. @araffin already reviewed all sections. I think we don't need to assign.
I am planning to submit it at the beginning of next month. So, any feedbacks are welcome until end of this month : ) .

AtsushiSakai, 2018-08-18:
@daniel-s-ingram @jwdinius @karanchawla @araffin @AlexisTM
By the way, this repository has reached 3000 stars finally🎉🎊. 
It takes almost 2 years from initial commit. 
I appreciate you guys so much🙌

AlexisTM, 2018-08-19:
3000 stars! This is a big achievement and acknowledgement of the usefulness of your repo :) 

I would add in the future work: a comparison table to find which algorithm fits the best

AtsushiSakai, 2018-08-19:
@AlexisTM Thank you : )

> I would add in the future work: a comparison table to find which algorithm fits the best for a specific application
Good idea. I will add it to the paper.

jwdinius, 2018-08-19:
I have created a PR in the paper repo with some suggested revisions to the paper.  Hopefully these will help you!  I am happy to go through the rest of the paper if my comments would help.

AtsushiSakai, 2018-08-20:
@jwdinius Thank you so much for your PR!!. It must help me. I will check it later. Thanks.

AtsushiSakai, 2018-08-20:
@karanchawla Thank you so much for your comment.
> I think we should add more in terms of related work

This is a good idea. If you know good related works, let me know.

> providing some scientific analysis for those for example running times, memory and so on.

I think the code of this repo is for education, not practical, so performance evaluation is not needed.

AtsushiSakai, 2018-08-24:
@daniel-s-ingram Thank you!! https://github.com/AtsushiSakai/PythonRoboticsPaper/issues/3

jwdinius, 2018-08-27:
@AtsushiSakai I will review the remainder of the paper by Wednesday of this week.  Cheers!

AtsushiSakai, 2018-08-27:
@jwdinius Thank you!!

AtsushiSakai, 2018-09-03:
@daniel-s-ingram @jwdinius @karanchawla @araffin @AlexisTM

I finally submitted the paper!!
You can check the paper here: http://arxiv.org/abs/1808.10703

I appreciate your review for this paper.
Thanks!!

closed 2018-09-03 by AtsushiSakai