AtsushiSakai/PythonRobotics #99: Html doc with read the docs ?

araffin, 2018-09-03:
Hi,

As the README is now huge, I was wondering if you would be interested by a html documentation using sphinx (and maybe read the docs).

I recently did that for one of my projects, and it makes things cleaner ;)
Example: http://stable-baselines.readthedocs.io/

shubh-agrawal, 2018-09-03:
Hey, I had an observation related to long readme with many GIFs. If I allow all the GIFs to load while scrolling down the readme, my browser starts using up a lot of memory. I use 8GB RAM pc and it certainly freezes after all GIFs are loaded by the browser. Can you verify this on your PC as well? If it goes same for you then I think we definitely need some replacement to long readme.

araffin, 2018-09-03:
I don't have any freezing (using firefox on linux with 8 GB RAM), but looking at the web inspector, the README loads 361,75 Mo, which is quite a lot.

Another advantage of html doc: it get referenced by web crawler like Google, which means more visibility ;)

AtsushiSakai, 2018-09-07:
@araffin Oh that's good. Your PR is welcome!!

araffin, 2018-09-07:
Ok, i'll try to do that this month