AtsushiSakai/PythonRobotics #101: code change

sean-ascii, 2018-09-14:
Hi,
      In this program, tcost always qeuals to node.cost in line 105, so line 108 and 109 will never run.
      Should line 105 to line 109 change to some code like this:
      `
              if openset[n_id] > node.cost:
                    openset[n_id] = node
       `

      I don't know exactly what's you think, so if my thought is incorrect, please let me know, thanks!

https://github.com/AtsushiSakai/PythonRobotics/blob/8f3023816fccd029506a1fe79e143dc4f4509687/PathPlanning/AStar/a_star.py#L105

AtsushiSakai, 2018-09-14:
@sean-ascii Hi. Thank you for your question.
> tcost always qeuals to node.cost
No, it might be different, because tcost is new proposed cost, but node.cost was proposed before from another graph route. 
tcost will be equal when tcost is better than node.cost.

sean-ascii, 2018-09-15:
@AtsushiSakai 
Thank you for your explanation!
I‘m just starting to learn A* algorithm, so maybe I have something incorrect in my understanding for this question.
But, doesn't the line 91 compute the the new proposed cost? And, I also thought the heuristic value will just need to be computed in the line 67, and line 103 does not.
Maybe I should read the algorithm paper again.

In addition, thank you very much for creating this project! This project is very useful for beginner like me to study these robotic algorithm, it is awesome!