AtsushiSakai/PythonRobotics #136: Graph SLAM duplicated pose 0 and virtual measurement angle error issue

ZhenghaoFei, 2018-11-24:
Hi Atsushi,

I am learning to write Graph SLAM these days, your codes helped me a lot, thank you very much.
I find two things in your code though.

**1. Observation not matching** 
I changed the Qsim and Rsim to 0 to test the system.
```
Qsim = np.diag([0.2, np.deg2rad(1.0)])**2*0
Rsim = np.diag([0.1, np.deg2rad(10.0)])**2*0
```
However, you will see the SLAM output didn't really match the true trajectory even in this noise-free case. 
It is due to the duplicated pose 0 node. 

You append first pose node twice in hxDR the main function, and duplicated observation of this pose  in graph_based_slam using  
`zlist.insert(1, zlist[0])
`
I recommend to delete
`zlist.insert(1, zlist[0])
`
in 
`def graph_based_slam(x_init, hz):
`
and only append first pose once in hxDR in the main function. 

I tested, it will give you￼
￼ perfect fit when you use zero Qsim and Rsim and also better fit using normal noise.

**2. Virtual measurement angle error issue** 
Solve problem 1 will give you perfect fit visually, but the cost residue is still large.  I think it is due to the  angle error in virtual measurement. Since your observation here didn't measure the landmark's heading (consider it as a point), so your virtual measurement between poses should not have any heading information.  I simply changed the angle error term and Jacobian row 3 to 0. The problem should be solved.

Thanks for your great code again, it really helped to understanding Graph SLAM!

Best,
Zhenghao

AtsushiSakai, 2018-11-24:
@ZhenghaoFei Thank you again. Can you make a PR?

ZhenghaoFei, 2018-11-24:
Sure. I will clean my code and PR tomorrow. Thanks!

AtsushiSakai, 2018-11-25:
@ZhenghaoFei Thank you. If you have time, your next PR of the jupyter documentation about Graph SLAM is wellcome!! #77

closed 2018-11-25 by AtsushiSakai

ZhenghaoFei, 2018-11-25:
Thanks for your invitation!. I will check it out when I get time.