AtsushiSakai/PythonRobotics #150: cgmres_nmpc

MahdiehSadabadi, 2019-01-02:
Have you considered the path tracking problem in the cgmres_nmpc code? The cost function of the nonlinear MPC problem in your code does not include any terms related to (x-x_ref), (y-y_ref), etc.

AtsushiSakai, 2019-01-03:
@MahdiehSadabadi Hi. No, I haven't. But it is possible I think. I only have linear MPC path tracking implementation:  [PythonRobotics/model\_predictive\_speed\_and\_steer\_control\.py at master · AtsushiSakai/PythonRobotics](https://github.com/AtsushiSakai/PythonRobotics/blob/master/PathTracking/model_predictive_speed_and_steer_control/model_predictive_speed_and_steer_control.py)
If you are interested in it, your PR is wellcome : )

closed 2019-01-03 by AtsushiSakai

MahdiehSadabadi, 2019-01-03:
@AtsushiSakai Many thanks for your reply. I have already implemented the linear MPC for path tracking problems. I am trying to use nonlinear MPC. I also think it would be possible to extend your code (cgmres_nmpc) for path tracking by modifying the MPC cost function, the partial derivative of Hamiltonian w.r.t. states, etc.

AtsushiSakai, 2019-01-03:
@MahdiehSadabadi Oh. great. If you can share the code, feel free to make a PR.

MahdiehSadabadi, 2019-01-04:
Sure thing! Once I finish my code, I might share it here.