AtsushiSakai/PythonRobotics #161: [Stanley Controller] Incorrect lateral (crosstrack) error term

amirx96, 2019-01-23:
Think you might have an error when calculating the lateral or cross-track error. 

In here, you take the RMS distance from the front of the vehicle. 

https://github.com/AtsushiSakai/PythonRobotics/blob/bb8b8b33d44d6ce3a04b2c16cd3a97fa36189abf/PathTracking/stanley_controller/stanley_controller.py#L130-L141

You need to take the projection of this error along the front axle vector in the global frame like so:

``` python
   # Calc front axle position
    fx = state.x + L * np.cos(state.yaw)
    fy = state.y + L * np.sin(state.yaw)

    # Search nearest point index
    dx = [fx - icx for icx in cx]
    dy = [fy - icy for icy in cy]
    d = [np.sqrt(idx ** 2 + idy ** 2) for (idx, idy) in zip(dx, dy)]
    front_axle_vec = [-np.cos(state.yaw + np.pi/2), -np.sin(state.yaw + np.pi/2)]
    closest_error = min(d)
    target_idx = d.index(closest_error)
    error_front_axle = np.dot( [ dx[target_idx], dy[target_idx] ], front_axle_vec)

    # target_yaw = normalize_angle(np.arctan2(
    #     fy - cy[target_idx], fx - cx[target_idx]) - state.yaw)
    # if target_yaw > 0.0:
    #     error_front_axle = - error_front_axle
```
Note that setting the manual signing of the error term is no longer needed with this calculation

You can check the effects between these two versions of the code by setting the `ds` term in https://github.com/AtsushiSakai/PythonRobotics/blob/bb8b8b33d44d6ce3a04b2c16cd3a97fa36189abf/PathTracking/stanley_controller/stanley_controller.py#L152 to to 1 or 2.

AtsushiSakai, 2019-01-24:
@amirx96 Thank you for your report. Is it possible for your to make a PR?

cd4806088, 2020-05-26:
> Think you might have an error when calculating the lateral or cross-track error.
>
> In here, you take the RMS distance from the front of the vehicle. [...]

can you talk about the projection? Thank you!

amirx96, 2020-05-26:
@cd4806088 

You want the error term for the stanley controller to be the lateral offset between the vehicle and closest point on the path. 

Previously, the error was just the magnitude of the position vector from the front axle to the closest point on the path. 

Imagine that if you are trying to follow a straight path with waypoints separated by 1m. If you are perfectly aligned to the path, then the next closest waypoint is up to 1 meter ahead. The magnitude of the position error vector is ~1 meter, but your lateral offset should be 0 if you are aligned.

Hence, you take a projection of the position error vector onto the lateral axis of the vehicle to obtain the lateral component of the error. In the straight waypoint path case, the lateral error would be zero and the stanley controller would not give a steering input when we are aligned to the path perfectly.

khaschuluu, 2025-01-28:
Project d to perpendicular of path (tangent of closest point) is sounds more right to me. Because heading error will handle where front axle should heading.