AtsushiSakai/PythonRobotics #167: BIT* not finding shorter path

ahq1993, 2019-02-04:
Hi,

Thank you for maintaining this awesome library. I am running your Batch Informed Trees (BIT*) implementation and it successfully finds an initial path. However, if I keep running for even very large number of iterations, it never reduces the length of that initial path. Is there anything I am missing to fix this?

AtsushiSakai, 2019-02-04:
@ahq1993 Hi. Thank you for your report. I'm sorry I'm not familiar with it.
@karanchawla Could you help him?

karanchawla, 2019-02-09:
Hey @ahq1993 can you share the grid that you're running on? I'll try to debug on my free time. 

Thanks @atsushi-sakai

AtsushiSakai, 2019-02-22:
@ahq1993 I will close this issue. If you need more help, please reopen it.

closed 2019-02-22 by AtsushiSakai