AtsushiSakai/PythonRobotics #174: appveyor test is broken

AtsushiSakai, 2019-03-03:
python setup.py imports numpy while having it as a dependency · Issue #85 · cvxgrp/scs https://github.com/cvxgrp/scs/issues/85

jwdinius, 2019-03-06:
It looks like the travis test is using python v3.6.8, but the appveyor test is using 3.6.5.  Is it possible to make these tests use the _same_ underlying conda environment (same python version and same package versions)?  It seems like a non-trivial amount of work is going into maintaining this appveyor ci.  My thought is that maybe the workload could be made lighter by just making sure the two tests are as similar as possible.

AtsushiSakai, 2019-03-07:
@jwdinius Thank you for your advice. That's good idea. Do you have any idea to achieve that?

AtsushiSakai, 2019-03-07:
@jwdinius One idea is that I make travis-ci and appveyor use a same enviroment.yml. And then, I set a specific version for each library.

AtsushiSakai, 2019-03-16:
I fixed it to pass the test which using cvxpy.

closed 2019-03-16 by AtsushiSakai