AtsushiSakai/PythonRobotics #181: Look-ahead distance in Pure Pursuit

Kitsunow, 2019-04-08:
Hi,

i'm not sure about it, but as far as i understand pure pursuit control, you search for points on a line until you reach the end of the look-ahead distance beginning from the car.

In your pure pursuit implementation though, you calculate the distance between points on the track and compare them to the look-ahead distance:

 ```
# search look ahead target point index
while Lf > L and (ind + 1) < len(cx):
      dx = cx[ind + 1] - cx[ind]
      dy = cy[ind + 1] - cy[ind]
      L += math.sqrt(dx ** 2 + dy ** 2)
      ind += 1
```

But shouldn't you compare each point to the car's position instead?
Like:

# search look ahead target point index
while Lf > L and (ind + 1) < len(cx):
      dx = cx[ind] - state.x
      dy = cy[ind] - state.y
      L += math.sqrt(dx ** 2 + dy ** 2)
      ind += 1
```

Kitsunow, 2019-04-08:
After reviewing Amidi1991 for the PP-Approach, i found the error on my site, since it states that the look-ahead is applied between the waypoints, not between car and points. Sorry!

zhengzh, 2019-04-10:
I think you are right, I checked the implementation of this [repo](https://github.com/larics/pure_pursuit/blob/00497f136cb702d8c55a2271090b4ac55c566e93/src/pure_pursuit.cpp#L151)

AtsushiSakai, 2019-04-10:
@Kitsunow @zhengzh Hi. Thank you for your reports. Oh you are right. According to this paper, https://www.ri.cmu.edu/pub_files/2009/2/Automatic_Steering_Methods_for_Autonomous_Automobile_Path_Tracking.pdf 
The distance from the robot position, not nearest path point. I will fix it.

AtsushiSakai, 2019-04-10:
[image]

AtsushiSakai, 2019-04-10:
I fixed it.

closed 2019-04-10 by AtsushiSakai