AtsushiSakai/PythonRobotics #183: LQR implementaion Issue

akhil22, 2019-04-13:
Hi,
I was going through your implementation of LQR for path and speed control : [code](https://github.com/AtsushiSakai/PythonRobotics/blob/master/PathTracking/lqr_speed_steer_control/lqr_speed_steer_control.py
)

AtsushiSakai, 2019-04-13:
@akhil22 Hi, in my understanding, LQR is only used for linear system. The dynamic nonlinear motion system is linearized around current state. see: https://github.com/AtsushiSakai/PythonRobotics/blob/ac066ee6049adc6315a859127344ee5de1e48b30/PathTracking/lqr_speed_steer_control/lqr_speed_steer_control.py#L102-L113 

So, Ak, Bk, Qk and Rk are not const, these are updated at every time step.  

> but the algorithm seems to do well in all of the cases.
That is Cool!!

closed 2019-04-13 by AtsushiSakai

akhil22, 2019-04-13:
> Hi,
> I was going through your implementation of LQR for path and speed control : [code](https://github.com/AtsushiSakai/PythonRobotics/blob/master/PathTracking/lqr_speed_steer_control/lqr_speed_steer_control.py)
> and one thing i noticed is that you are solving the DARE equation under the assumption that matrices Ak,Bk, Qk and Rk are constants (K will converge if this assumption is true) but since your dynamics are non linear these matrices are not constant. Your code is still working, am I missing something here ? or the code is working since the robot position is always close to the local goal and because of that the linear model works reasonably well(assumption that Ak, and Bk are constant is valid)
> 
> I tried with different initial positions which are far from the trajectory but the algorithm seems to do well in all of the cases.

Never mind in the code Ak and Bk are constants for most cases except when the 

> @akhil22 Hi, in my understanding, LQR is only used for linear system. The dynamic nonlinear motion system is linearized around current state. see:
> 
> [PythonRobotics/PathTracking/lqr_speed_steer_control/lqr_speed_steer_control.py](https://github.com/AtsushiSakai/PythonRobotics/blob/ac066ee6049adc6315a859127344ee5de1e48b30/PathTracking/lqr_speed_steer_control/lqr_speed_steer_control.py#L102-L113)
> 
> Lines 102 to 113 in [ac066ee](/AtsushiSakai/PythonRobotics/commit/ac066ee6049adc6315a859127344ee5de1e48b30)
>  A = np.zeros((5, 5)) 
>  A[0, 0] = 1.0 
>  A[0, 1] = dt 
>  A[1, 2] = v 
>  A[2, 2] = 1.0 
>  A[2, 3] = dt 
>  A[4, 4] = 1.0 
>  # print(A) 
>   
>  B = np.zeros((5, 2)) 
>  B[3, 0] = v / L 
>  B[4, 1] = dt 
> 
> So, Ak, Bk, Qk and Rk are not const, these are updated at every time step.
> 
> > but the algorithm seems to do well in all of the cases.
> > That is Cool!!
Yeah they are not constant and hence using the convergence method to solve dare seems to be invalid but the only parameter changing in Ak and Bk is the target velocity which is constant except when the vehicle is changing its direction which is rare and hence it seems to be working ..