AtsushiSakai/PythonRobotics #185: Taking rear axis instead of center for PurePursuit

Kitsunow, 2019-05-09:
In your pure pursuit example you're using the center of the car for calculation, but in the tech report from Craig Coulter and in a research paper form Jarrod Snider the rear axle is used.

I found significant better behaviour when implementing it with this variant in the physical world, thus asking you if you could change this in your example.

Instead of using `state.x` directly you could modify the state by adding the rear axle as 
```
offset_x = (L / 2) * math.cos(state.yaw)
offset_y = (L / 2) * math.sin(state.yaw)
state.rear_x = state.x - offset_x
state.rear_y = state.y - offset_y
```
and use `state.rear_x` and `state.rear_y` instead of the center now.

AtsushiSakai, 2019-05-09:
@Kitsunow Hi. Thank you for your suggestion!! Can you make a PR for it?

Kitsunow, 2019-05-09:
I created a PR.

closed 2019-05-10 by AtsushiSakai