AtsushiSakai/PythonRobotics #186: Runtime optimization for PurePursuit

Kitsunow, 2019-05-09:
In the `calc_target_index` function you always calculate the distance between all points on the track. I think this is unnecessary, you could optimize this by calculating it once and remember the last target_index, thus only recalculating it for a short range next time:

```
old_target_index = None

def calculate_distance(state, point_x, point_y):
        distance_x = state.rear_x - point_x
        distance_y = state.rear_y - point_y
        return math.sqrt(distance_x ** 2 + distance_y ** 2)

def calc_target_index(state, cx, cy):
    if old_target_index is None:
        # calculate the distances like before, just do the following at the end:
        old_target_index = ind
    else:
        while True:
                distance_this_index = calculate_distance(state, cx[index], cy[index])
                index = index + 1 if (index + 1) < len(path_x) else 0
                distance_next_index = calculate_distance(state, cx[index], cy[index])
                if distance_this_index < distance_next_index:
                    break
                old_target_index = index
```

This way my code only used  1/10 runtime :)

AtsushiSakai, 2019-05-09:
@Kitsunow Hi. Thank you for your suggestion!! Can you make a PR for it?

Kitsunow, 2019-05-09:
I created a PR.

closed 2019-05-10 by AtsushiSakai