AtsushiSakai/PythonRobotics #200: Particle filter occasionally dies because pw.sum == 0

kjaget, 2019-06-07:
'''
particle_filter.py:126: RuntimeWarning: invalid value encountered in divide
  pw = pw / pw.sum()  # normalize
Traceback (most recent call last):
particle_filter.py start!!
  File "particle_filter.py", line 254, in <module>
    main()
  File "particle_filter.py", line 247, in main
    plot_covariance_ellipse(xEst, PEst)
  File "particle_filter.py", line 162, in plot_covariance_ellipse
    eigval, eigvec = np.linalg.eig(Pxy)
  File "/usr/local/lib/python2.7/dist-packages/numpy/linalg/linalg.py", line 1143, in eig
    _assertFinite(a)
  File "/usr/local/lib/python2.7/dist-packages/numpy/linalg/linalg.py", line 216, in _assertFinite
    raise LinAlgError("Array must not contain infs or NaNs")
numpy.linalg.linalg.LinAlgError: Array must not contain infs or NaNs
```
Seems more likely to happen if NP is increased, but this does happen even if the default value is used.

From some basic debugging, it looks to get in a state where all px entries are the same. If those happen to be far enough away from the ground truth location, all the weights end up as 0

When this happens, it looks like the resample code always picks the same index.  I'm not sure if that is a cause or an effect, though.

kjaget, 2019-06-08:
Update - this seems to only be a problem with python2. Running with python2, the base array in resampling is all zeros and inds contains all the same index.  With python3, it has non-zero values, the inds array indexes many different points, and the code seems to run correctly.

I updated numpy to the latest for python2 but it did not correct the problem.

So I have a workaround, but wanted to give more info in case this is something you'd like to fix.

AtsushiSakai, 2019-06-08:
@kjaget Hi. I'm sorry I am not using python2 anymore. As you can see the README.md, this repo is only focusing python3.

closed 2019-06-09 by AtsushiSakai