AtsushiSakai/PythonRobotics #208: Rewiring tree in RRT* algorithms

peweetheman, 2019-07-03:
Rewiring the tree in all of the RRT* algorithms (and maybe others) the costs of children nodes are never updated

Example, if you have:

A->B->C->D

Then suppose node E is added as a child of A which finds shorter path from A->C, making tree be rewired such that:

A->E->C->D
A->B

Now node C cost would be adjusted, however node D cost should be lowered but is not.
It makes the most sense to me to have costs of nodes that were lowered be propagated to children like in Dijkstra's, A*, or many other search algos.

Is this an issue with the implementation or really how the RRT* algorithm runs?

AtsushiSakai, 2019-07-03:
@peweetheman Thank you for your report. Can you make a PR to fix it?

AtsushiSakai, 2019-07-14:
@peweetheman Hi. I fixed the issue, could you please check the issue is solved or not?
https://github.com/AtsushiSakai/PythonRobotics/blob/master/PathPlanning/RRTStar/rrt_star.py

AtsushiSakai, 2019-07-15:
Fix list
- [ ] RRT
- [ ] RRTStar
- [ ] BatchInformedRRTStar
- [ ] ClosedLoopRRTStar
- [ ] InformedRRTStar
- [ ] LQRRRTStar
- [ ] RRTDubins
- [ ] RRTStarDubins
- [ ] RRTStarReedsShepp

peweetheman, 2019-07-19:
Yes this solved the issue. 

The propagate_cost_to_leaves() definitely could be made faster if you track children as well, which is very little overhead, and pretty easy to implement (just wherever you set node.parent add the node to the children list)

Currently O(n^2) I believe, and would be O(n) if you just iterate through children and then recurse as below. 

 def propagate_cost_to_leaves(self, parent_node):
        for node in parent_node.children
                d, _ = self.calc_distance_and_angle(parent_node, node)
                node.cost = parent_node.cost + d
                self.propagate_cost_to_leaves(node)

Thanks!

closed 2019-08-03 by AtsushiSakai