AtsushiSakai/PythonRobotics #211: TypeError: 'int' object is not iterable

jash101, 2019-07-10:
I am not able to run Grid Based Sweep Path Planning.
Following is the terminal output:

```
start!!
Traceback (most recent call last):
  File "grid_based_sweep_coverage_path_planner.py", line 307, in <module>
    main()
  File "grid_based_sweep_coverage_path_planner.py", line 289, in main
    planning_animation(ox, oy, reso)
  File "grid_based_sweep_coverage_path_planner.py", line 262, in planning_animation
    px, py = planning(ox, oy, reso)
  File "grid_based_sweep_coverage_path_planner.py", line 248, in planning
    gmap, xinds_goaly, goaly = setup_grid_map(rox, roy, reso, sweeping_direction)
  File "grid_based_sweep_coverage_path_planner.py", line 194, in setup_grid_map
    xinds_goaly = [[]], goaly = 0
TypeError: 'int' object is not iterable
```

AtsushiSakai, 2019-07-10:
@jash101 Hi. Thank you for your report? Did you change anything in the code?

jash101, 2019-07-10:
No. Just installed dependencies, cloned the repository and ran the code

AtsushiSakai, 2019-07-10:
@jash101 OK. weird. What is your Python version?

jash101, 2019-07-10:
Python 3.5.2 Can that be the problem? I'm having trouble installing cvxpy on python 3.6.9, so tried running your package on 3.5. Mapping codes work fine

AtsushiSakai, 2019-07-10:
@jash101 If you use Python 2.x, it might be problem. Let me check the latest code.

AtsushiSakai, 2019-07-10:
@jash101 I found my commit mistake. Could you please try latest code?

jash101, 2019-07-10:
Sure. I'll clone it again

jash101, 2019-07-10:
Works fine now :+1:

closed 2019-07-10 by jash101

AtsushiSakai, 2019-07-10:
@jash101 Thank you so much for your report!! I added an unittest for CI.