AtsushiSakai/PythonRobotics #217: Wrong obstacle size in rrt connect

tolstoy92, 2019-08-14:
Hi! I think, than in check_collision func we're need size*2!

```
    def check_collision(node, obstacleList):
        for (ox, oy, size) in obstacleList:
            dx_list = [x - ox for x in node.path_x]
            dy_list = [y - oy for y in node.path_y]
            d_list = [dx * dx + dy * dy for (dx, dy) in zip(dx_list, dy_list)]
            if min(d_list) <= (size*2) ** 2:  **< !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!**
                return False  # collision
        return True  # safe
```

AtsushiSakai, 2019-08-14:
@tolstoy92 Hi. Thank you for your comment. Why do we need size*2?. In the code, size means the radius of obstacle.

cloudflyMa, 2019-08-15:
python @  invalid syntax

AtsushiSakai, 2019-08-18:
@cloudflyMa I think your python or numpy is too old. Please check the version of it.

AtsushiSakai, 2019-08-18:
@tolstoy92 I want to close this issue once. If you have any idea about it. Please comment to this issue.

closed 2019-08-18 by AtsushiSakai