AtsushiSakai/PythonRobotics #224: Paths ignoring collision(s)...

alex-ira, 2019-09-16:
Hi Atsushi

Great work! I've just started exploring and playing but am getting results as in figure below. I'd expect to get "Cannot find path". Am I missing something? 

Thanks in advance. 

Cheers
[image]

AtsushiSakai, 2019-09-16:
@alex-ira Thank you for your bug report!!. Did you use latest code? I fix a bug for it: https://github.com/AtsushiSakai/PythonRobotics/issues/223

alex-ira, 2019-09-16:
@AtsushiSakai Yes, I did. Thanks for a quick reply!

AtsushiSakai, 2019-09-16:
@alex-ira Currently, default path resolution is 1 meter, so the collision check might skip the obstacle. 
https://github.com/AtsushiSakai/PythonRobotics/blob/8eef2fd7bbda218d2c381c4a29a17c1f6f710591/PathPlanning/RRT/rrt.py#L35

Could you please check the same problem happen when you set the path_reolusion = 0.1?

alex-ira, 2019-09-16:
@AtsushiSakai I think that might be the issue. I've tried path_resolution = 0.1 and got this in one of the trials:
[image]

alex-ira, 2019-09-16:
@AtsushiSakai this one is with path_resolution = 0.05:
[image]

alex-ira, 2019-09-16:
@AtsushiSakai I just observed the same issue with RRTStar (see below; I use path_resolution = 0.1).
[image]

alex-ira, 2019-09-16:
@AtsushiSakai the issue seems to go away if I reduce expand_dis.

AtsushiSakai, 2019-10-01:
@alex-ira Sorry for late reply. Could you please try the latest rrt.py code? It might from circle visualization problem. Thanks for your report.

alex-ira, 2019-10-01:
@AtsushiSakai thanks for getting back to me. I just tried it and unfortunately there are still the same  issues (see pictures below). 

Expand_dis = 3, path_resolution = 3
[image]

Expand_dis = 3, path_resolution = 1
[image]

Expand_dis = 3, path_resolution = 0.1
[image]

Expand_dis = 5, path_resolution = 0.1
[image]

I've been also playing with RRTStar have the same issues. 

Thanks for your time in advance. Cheers

alex-ira, 2019-10-02:
@AtsushiSakai might be relevant for debugging. The picture below is for RRTStar with expand_dis = 5 and path_resolution = 1:
[image]

AtsushiSakai, 2019-10-02:
@alex-ira Oh, Obviously the last path expansion has problem. Thank you for your report. I will try to fix it.

AtsushiSakai, 2019-10-02:
@alex-ira I fixed it. Could you please check it again?

alex-ira, 2019-10-02:
@AtsushiSakai thanks for your work. RRT seems to be okay, however I'm finding the same issue with RRTStar. 

expand_dis = 5, path_resolution = 0.5:

[image]

[image]

AtsushiSakai, 2019-10-03:
@alex-ira Are you using latest rrt_star.py? Your figure looks using old code.

[image]

alex-ira, 2019-10-03:
@AtsushiSakai no, I'm using the latesdt rrt_star.py. I just kept `"-xg"` to observe path_resolution.

alex-ira, 2019-10-04:
@AtsushiSakai here is one with the new visuals.

[image]

AtsushiSakai, 2019-10-05:
@alex-ira Oh, thank you for your kind report. I fixed an another bug on rrt_star.py. Could you try it again? Sorry.

alex-ira, 2019-10-07:
@AtsushiSakai thanks for all the work! It appears to be ok now.

closed 2019-10-07 by alex-ira

AtsushiSakai, 2019-10-07:
@alex-ira Thank you so much for you effort. I appreciate it.