AtsushiSakai/PythonRobotics #237: Why is the index value calculated?

lining9717, 2019-10-12:
In Dijkstra's algorithm, the index value of the node may be repeated. 
Why is the index value calculated?
`def calc_index(self, node):
        return (node.y - self.miny) * self.xwidth + (node.x - self.minx)`

AtsushiSakai, 2019-10-12:
@lining9717 I'm sorry, I cannot understand you question. Could you please explain for details?

lining9717, 2019-10-12:
@AtsushiSakai I mean why you calculate the index value of each point like this
`    def calc_index(self, node):
        return (node.y - self.miny) * self.xwidth + (node.x - self.minx)`，because I think there will be some duplicate indexes.

AtsushiSakai, 2019-10-12:
@lining9717 I'm sorry I don't understand your question... sorry.

closed 2019-10-12 by AtsushiSakai