AtsushiSakai/PythonRobotics #251: k-means clustering: ncluster value other than 2

RajeshKumarS, 2019-11-21:
Setting ncluster to greater than 2 (tried with 3 & 4) gives the following error:

IndexError: arrays used as indices must be of integer (or boolean) type.

Following is the full dump:

Time: 0.0
Traceback (most recent call last):
  File "d:/apps/PythonRobotics-master/Mapping/kmeans_clustering/kmeans_clustering.py", line 179, in <module>
    main()
  File "d:/apps/PythonRobotics-master/Mapping/kmeans_clustering/kmeans_clustering.py", line 161, in main
    clusters = kmeans_clustering(rx, ry, ncluster)
  File "d:/apps/PythonRobotics-master/Mapping/kmeans_clustering/kmeans_clustering.py", line 41, in kmeans_clustering
    clusters = calc_centroid(clusters)
  File "d:/apps/PythonRobotics-master/Mapping/kmeans_clustering/kmeans_clustering.py", line 54, in calc_centroid
    x, y = calc_labeled_points(ic, clusters)
  File "d:/apps/PythonRobotics-master/Mapping/kmeans_clustering/kmeans_clustering.py", line 88, in calc_labeled_points
    x = tx[inds]
IndexError: arrays used as indices must be of integer (or boolean) type

Also, I still got two clusters even if ncluster was set to 1.

AtsushiSakai, 2019-12-07:
@RajeshKumarS Thank you for reporting. You are totally right. I fixed the code. If you encounter the problem again, please reopen or make a new issue. Thank!!

closed 2019-12-07 by AtsushiSakai

RajeshKumarS, 2019-12-09:
@AtsushiSakai Thanks. It works fine and am able to use it without any problem. Thanks again.