AtsushiSakai/PythonRobotics #264: two questions in code 'graphSLAM_doc.ipynb'

zhchyang2004, 2019-12-24:
Hi, AtsushiSakai, your code really benefits robotics learners. Thank you!

When I study your code in 'graphSLAM_doc.ipynb', I find two questions to be discussed with you as follows.

1.  In 'Minimal Example' ,  do you think 'omegas' should be added onto the calculation of b?
        b[t1] += omegas * measure_constraints[(t1,t2)]
        b[t2] -=  omegas * measure_constraints[(t1,t2)]

2. In [6] ,  yaw1 and yaw2 has been included in tangle1 and tangle2.  So  'yaw1 and yaw2' is added twice in 'calculate the error of the virtual measurement'. 

      edge.e[2, 0] = pi_2_pi(yaw2 - yaw1 - tangle1 + tangle2)

      Should the correct code be like this?
      edge.e[2, 0] = pi_2_pi(- tangle1 + tangle2) 

Thank you for your kind feedback!

AtsushiSakai, 2019-12-27:
@zhchyang2004 Hi. I'm sorry for late reply.
@adelizer Could you please answer his question?

adelizer, 2020-01-07:
Hi, I will look into your remarks

AtsushiSakai, 2020-01-28:
@adelizer How about your progress?

adelizer, 2020-02-03:
Regarding remark [1] It is correct. Indeed we need to add the measurement noise Q into the calculation. It was not showing in the results as this was a minimal example. 
I am having a problem reproducing the bad results in remark [2] but I will work on a new PR with the fix.

AtsushiSakai, 2020-02-04:
@adelizer Thank you for check it.
>It is correct
OK I fixed it in #283
>but I will work on a new PR with the fix.
OK. great. You PR is wellcome.

closed 2020-05-16 by AtsushiSakai