AtsushiSakai/PythonRobotics #276: Dubin's goal doesn't end where expected

tsturzl, 2020-01-24:
**Describe the bug**
DubinsPath goal has a margin of error, seems to be off by 0.2m

**Expected behavior**
DubinsPath last point is equal to the goal

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Python 3.8
 - Latest as of posting

tsturzl, 2020-01-24:
I've ported the dubins path implementation to another language and I'm seeing the same issue. I'm wondering if there's a reason for this, and if this is expected behavior. I've looked at RRT Dubins and I noticed you're just moving the RRT nodes to where the dubins path ends. If some explanation can be given as to why this is the case I'd be happy to try and submit a PR if that is helpful.

tsturzl, 2020-01-24:
[https://i.postimg.cc/pLw731J3/dubins-loop.png](url) this is what I'm seeing. I'm just going from point a to point b and then looping back from point b to point a, I'd expect this to make a closed loop, but the end of each path is not exactly located on the goal.

AtsushiSakai, 2020-01-27:
@tsturzl Hi. Thank you for your report. I believe that is because of accumulated error of  generate_course function: https://github.com/AtsushiSakai/PythonRobotics/blob/123c4f777ccfc49a0a11a8b99c9a63013d46b9a7/PathPlanning/DubinsPath/dubins_path_planning.py#L214

AtsushiSakai, 2020-01-27:
I will fix this issue on #277

AtsushiSakai, 2020-01-28:
@tsturzl Hi. I believe I fixed this issue. If you find this bug again, please reopen it. Thanks.

closed 2020-01-28 by AtsushiSakai

tsturzl, 2020-01-31:
I'll take a look at your changes. A lot of your work has been really helpful, and I've been using it as a reference for a few of my own implementations. I appreciate the work you do.