AtsushiSakai/PythonRobotics #285: RRT star rewire() not work

sldai, 2020-02-06:
**Describe the bug**
In RRTStar class, 

    def rewire(self, new_node, near_inds):
        for i in near_inds:
            near_node = self.node_list[i]
            edge_node = self.steer(new_node, near_node)
            if not edge_node:
                continue
            edge_node.cost = self.calc_new_cost(new_node, near_node)

            no_collision = self.check_collision(edge_node, self.obstacle_list)
            improved_cost = near_node.cost > edge_node.cost

            if no_collision and improved_cost:
                near_node = edge_node                    **near_node now points to a new node not in the tree**
                near_node.parent = new_node          **The parent of nodes in the tree does not actually change **
                self.propagate_cost_to_leaves(new_node)

AtsushiSakai, 2020-02-07:
@sldai Can you create a PR?

sldai, 2020-02-09:
Yeah, I will try to do it.

sldai, 2020-02-09:
Hi, AtsushiSakai. 
In the original RRT* paper, the vertices near the sample are checked in a range of min(expand_dist, connect_circle_dist).
In your implementation:  r = self.connect_circle_dist * math.sqrt((math.log(nnode) / nnode))
I add: r = min(r, self.expand_dis)

This guarantees the neighbor vertices are accessible from the new sample. But a problem exists that the rewire process becomes myopic, so much more samples are needed to get a optimal path. This is also shown in the RRT* paper. 

Have you ever seen any paper solving this problem?

AtsushiSakai, 2020-02-09:
@sldai I'm sorry I don't know it..

PR #291 (2020-02-10): fix RRT rewire
fix #285
closed 2020-02-10 by AtsushiSakai