AtsushiSakai/PythonRobotics #293: Cannot chose reso value freely in the coverage path planner

KobeLaBryant, 2020-02-13:
Hi,
first off your code is exactly what im looking for but i could use a tip or two.
Im trying to use the grid_based_sweep_coverage_path_planner.py. From my understanding the reso-value dictates the distance between the paths. But i cannot freely set the value. 

I i e.g. use 
    ox = [0.0, 50.0, 50.0, 0.0, 0.0]
    oy = [0.0, 0.0, 50.0, 50.0, 0.0]
    
    reso = 0.5

the script throws an exception. In before i dig into the code i thought i might just ask here.

I have Python 3.7.4 and cloned the repo this morning.

Thank you

AtsushiSakai, 2020-02-15:
@KobeLaBryant Hi. Thank you for your reports. Could you please paste the exception log?

KobeLaBryant, 2020-02-17:
Hi,

Exception has occurred: IndexError
list index out of range
  File "C:\Users\eike.spielbrink\Desktop\PythonRobotics-master\Mapping\grid_map_lib\grid_map_lib.py", line 55, in get_value_from_xy_index
    return self.data[grid_ind]
  File "C:\Users\eike.spielbrink\Desktop\PythonRobotics-master\Mapping\grid_map_lib\grid_map_lib.py", line 164, in check_occupied_from_xy_index
    val = self.get_value_from_xy_index(xind, yind)
  File "C:\Users\eike.spielbrink\Desktop\PythonRobotics-master\PathPlanning\GridBasedSweepCPP\grid_based_sweep_coverage_path_planner.py", line 61, in move_target_grid
    while not gmap.check_occupied_from_xy_index(ncxind + self.moving_direction, ncyind, occupied_val=0.5):
  File "C:\Users\eike.spielbrink\Desktop\PythonRobotics-master\PathPlanning\GridBasedSweepCPP\grid_based_sweep_coverage_path_planner.py", line 223, in sweep_path_search
    cxind, cyind = sweep_searcher.move_target_grid(cxind, cyind, gmap)
  File "C:\Users\eike.spielbrink\Desktop\PythonRobotics-master\PathPlanning\GridBasedSweepCPP\grid_based_sweep_coverage_path_planner.py", line 258, in planning
    px, py = sweep_path_search(sweep_searcher, gmap)
  File "C:\Users\eike.spielbrink\Desktop\PythonRobotics-master\PathPlanning\GridBasedSweepCPP\grid_based_sweep_coverage_path_planner.py", line 268, in planning_animation
    px, py = planning(ox, oy, reso)
  File "C:\Users\eike.spielbrink\Desktop\PythonRobotics-master\PathPlanning\GridBasedSweepCPP\grid_based_sweep_coverage_path_planner.py", line 298, in main
    planning_animation(ox, oy, reso)
  File "C:\Users\eike.spielbrink\Desktop\PythonRobotics-master\PathPlanning\GridBasedSweepCPP\grid_based_sweep_coverage_path_planner.py", line 321, in <module>
    main()

AtsushiSakai, 2020-02-22:
@KobeLaBryant I think your problem has been fixed by #295. 
Please check latest master code. 
Thank you for reporting!!

commit e8ffc01c (2020-02-22): Merge pull request #295 from AtsushiSakai/issue_293

fix #293
closed 2020-02-22 by AtsushiSakai