AtsushiSakai/PythonRobotics #296: Graph SLAM contribution

JeffLIrion, 2020-02-22:
I recently open sourced `graphslam`, which is a Graph SLAM solver written in Python ([GitHub](https://github.com/JeffLIrion/python-graphslam) / [PyPi](https://pypi.org/project/graphslam/)). Some of its features include:

* Optimize `R^2`, `R^3`, `SE(2)`, and `SE(3)` datasets
* Extensible: support for new manifolds and edge types can easily be added
* Minimal dependencies: only `numpy` and `scipy`
* Thorough [documentation](https://python-graphslam.readthedocs.io/en/stable/index.html), including a brief derivation of [Graph SLAM](https://python-graphslam.readthedocs.io/en/stable/graphslam.graph.html#module-graphslam.graph)
* Ease of use: you can literally go from zero to optimizing a graph in 4 commands

    ```python
    # 1. Install the package via `pip install graphslam`

    # 2. Import
    from graphslam.load import load_g2o_se3

    # 3. Load a graph
    g = load_g2o_se3("parking-garage.g2o")

    # 4. Optimize
    g.optimize()
    ```

I think this package would fit nicely with the aim of this project. Please let me know if you would be interested in such a contribution.

AtsushiSakai, 2020-02-23:
Thank you for your suggestion, and I love your OSS.
Any kind of your contribution is welcome : )
For example, adding new simulation or document based on your OSS.
However, adding your OSS as a dependency of this project is difficult because I want to keep the dependency as simple as possible.

JeffLIrion, 2020-02-23:
OK, I'll think about whether I could strip out some code from the package and use it to create an example for this project. A problem in `R^2` or `R^3` would probably be too simple, so maybe a good contribution would be visualizing the optimization of a small (~100 vertices) graph in `SE(2)` using numerical differentiation. 

How does that sound?

AtsushiSakai, 2020-02-23:
That great. Could you create a PR?

JeffLIrion, 2020-02-25:
I'll work on this when I get some time. What format would you prefer?

My thought is to have the Graph SLAM code in a Python module, and then in a Jupyter notebook I will document the Graph SLAM problem formulation, setup an example problem, and use the code in the Python module to optimize it.

AtsushiSakai, 2020-02-26:
I think you idea is perfect. Let's discuss the details in a PR.

closed 2020-03-25 by AtsushiSakai