AtsushiSakai/PythonRobotics #323: a problem in fast_slam1.py

zhchyang2004, 2020-05-01:
Hi, in 'add_new_lm', do you think Gz should be defined as Jacobian w.r.t the lamdmark's coordinate  as follows, instead of the range  and bearing? 

dx = r * c
dy = r * s
d2 = dx**2 + dy**2
d = math.sqrt(d2)
Gz = np.array([[dx / d, dy / d],
                        [-dy / d2, dx / d2]])
particle.lmP[2 * lm_id:2 * lm_id + 2] = np.linalg.inv(Gz) @ Q @ np.linalg.inv(Gz.T)

Thanks!

AtsushiSakai, 2020-05-28:
@zhchyang2004 Thank you for your comments. Could you please explain why do you think? Do you have any reference for it?

zhchyang2004, 2020-05-29:
Hi, @AtsushiSakai 

You can find reference from Line 18 Table13.2 in the book 'Probabilistic Robotics'.

And also from Page 12,15 and16 in Claus Brenner's material:  https://github.com/jfrascon/SLAM_AND_PATH_PLANNING_ALGORITHMS/blob/master/07-FAST_SLAM/NOTES/F1-Fast_SLAM.pdf

In the upper part Page 16, it says 'Now we have only the covariance of the observation Qz, that is only term we have in this situation'. 

When initializing a new landmark, we only have the covariance from observation appliance Q, and predicted landmark's world coordinate from range and bearing. So it is reasonable to get the initialized landmark 's covariance by this way.

Thanks!

PR #334 (2020-06-05): fix inaccurate covariance calculation to add a new landmark in FastSLAM1,2
fix #323
closed 2020-06-05 by AtsushiSakai