AtsushiSakai/PythonRobotics #331: xTrue and xEst in ekf-slam algorithms

remblim, 2020-05-25:
The real position is used to calculate the angle to the landmarks, but the estimated angle should be used. This is because the detections made by the robot sensor use the estimated angle compared to the real world instead of the real angle wich is unavaillable to the robot.

So, change this:
       angle = pi_2_pi(math.atan2(dy, dx) - xTrue[2, 0])
into:
       angle = pi_2_pi(math.atan2(dy, dx) - xEst[2, 0])

and sent the xEst into the method at line 227 with the method accepting it at line 69.

AtsushiSakai, 2020-05-28:
Thank you for your comments.
> the real angle wich is unavaillable to the robot.

Hmm, I don't think so. Because the observation come from actual angle difference but the robot's does not.

AtsushiSakai, 2020-06-04:
Once I close it.

closed 2020-06-04 by AtsushiSakai