AtsushiSakai/PythonRobotics #341: on reeds_shepp curve type in  reeds_shepp_path_planning.py

oridong, 2020-06-16:
why there are type SLS  from function  straight_left_straight(x, y, phi). I read the paper but this type doesn't exist. 
Does this code is implement the algorithm completely?

AtsushiSakai, 2020-06-22:
@oridong Hi. I don't remember it... maybe I used another paper or oss for implementation, not original paper.

>  maybe it lacks the last five base words in the picture
Yes, you are right. These should be implemented. Your PR is welcome : )

lentychang, 2020-06-22:
Just for extra info:

Following your references, I found from table in page 135 in [this dissertation](https://homepages.laas.fr/jpl/promotion/chap3.pdf),
they propose more effective way to categorize all 48 combinations since these combinations don't all appear at same time. They can be categorized based on the orientation of car and orientation of steering angle (if I understand correctly).

PR #949 (2024-01-08): Reopen: Added missing path types
fix #341 

Reopening the PR #892 since it was closed due to deletion of head repository.

Committed the same changes:
1. Added missing RS paths and refactored the code.
2. Modified assert condition in test3 of test_rrt_star_reeds_shepp.py.

Summarising the discussion from PR #892:
1. "x" in the function names refers to the direction switches.
2. Had to change the assert condition in test3 of test_rrt_star_reeds_shepp.py due to some strange behaviour. I have mentioned this in detail in  this [issue](https://github.com/AtsushiSakai/PythonRobotics/issues/944).

## [... cut]
closed 2024-01-08 by AtsushiSakai