AtsushiSakai/PythonRobotics #342: CVXPY cannot be installed

willzoe, 2020-06-17:
The development environment I currently use is Ubuntu16.04+Python3.8+anaconda+pycharm.

But when I install CVXPY in anaconda to the virtual environment, it prompts that it cannot be found.

What is the recommended development environment?

lentychang, 2020-06-18:
Hi @willzoe,  

try `conda install -c conda-forge cvxpy -n <ENV_NAME>` it should work, unless you meet some unsolved dependency.

or if conda doesn't work, you can just use pip to intall

AtsushiSakai, 2020-06-22:
@willzoe Hi please check install doc of cvxpy: https://www.cvxpy.org/install/
If it don't work, please report it for cvxpy team.

@lentychang  Hi. Thank you for your support. I applicate it.

closed 2020-06-22 by AtsushiSakai