AtsushiSakai/PythonRobotics #364: bread first search.py AttributeError: 'Node' object has no attribute 'parent_index'

UncleBob2, 2020-07-18:
C:/Python38/bread first search.py start!!
min_x: -10
min_y: -10
max_x: 60
max_y: 60
x_width: 35
y_width: 35
Find goal
Traceback (most recent call last):
  File "C:/Python38/bread first search.py", line 250, in <module>
    main()
  File "C:/Python38/bread first search.py", line 241, in main
    rx, ry = bfs.planning(sx, sy, gx, gy)
  File "C:/Python38/bread first search.py", line 107, in planning
    rx, ry = self.calc_final_path(ngoal, closed_set)
  File "C:/Python38/bread first search.py", line 114, in calc_final_path
    n = closedset[ngoal.parent_index]
AttributeError: 'Node' object has no attribute 'parent_index'

AtsushiSakai, 2020-08-02:
@UncleBob2 Hi. Thank you for reporting. Could you give me entire your code to reproduce the issue?

UncleBob2, 2020-08-03:
PS C:\Users\Newtonian> & C:/Python38/python.exe "c:/Python38/bread first search.py"c:/Python38/bread first search.py start!!min_x: -10min_y: -10max_x: 60max_y: 60x_width: 35y_width: 35Find goalTraceback (most recent call last):  File "c:/Python38/bread first search.py", line 250, in <module>    main()  File "c:/Python38/bread first search.py", line 241, in main    rx, ry = bfs.planning(sx, sy, gx, gy)  File "c:/Python38/bread first search.py", line 107, in planning    rx, ry = self.calc_final_path(ngoal, closed_set)  File "c:/Python38/bread first search.py", line 114, in calc_final_path    n = closedset[ngoal.parent_index]AttributeError: 'Node' object has no attribute 'parent_index'

Kind regards,
Winsor

 
 

  
Breadth-First grid planning
author: Erwin Lejeune (@spida_rwin)
See Wikipedia article (https://en.wikipedia.org/wiki/Breadth-first_search)
"""

import math

import matplotlib.pyplot as plt

show_animation = True

class BreadthFirstSearchPlanner:

    def __init__(self, ox, oy, reso, rr):
        """
        Initialize grid map for bfs planning
        ox: x position list of Obstacles [m]
        oy: y position list of Obstacles [m]
        resolution: grid resolution [m]
        rr: robot radius[m]
        """

        self.reso = reso
        self.rr = rr
        self.calc_obstacle_map(ox, oy)
        self.motion = self.get_motion_model()

    class Node:
        def __init__(self, x, y, cost, pind, parent):
            self.x = x  # index of grid
            self.y = y  # index of grid
            self.cost = cost
            self.pind = pind
            self.parent = parent

        def __str__(self):
            return str(self.x) + "," + str(self.y) + "," + str(
                self.cost) + "," + str(self.pind)

    def planning(self, sx, sy, gx, gy):
        """
        Breadth First search based planning
        input:
            s_x: start x position [m]
            s_y: start y position [m]
            gx: goal x position [m]
            gy: goal y position [m]
        output:
            rx: x position list of the final path
            ry: y position list of the final path
        """

        nstart = self.Node(self.calc_xyindex(sx, self.minx),
                           self.calc_xyindex(sy, self.miny), 0.0, -1, None)
        ngoal = self.Node(self.calc_xyindex(gx, self.minx),
                          self.calc_xyindex(gy, self.miny), 0.0, -1, None)

        open_set, closed_set = dict(), dict()
        open_set[self.calc_grid_index(nstart)] = nstart

        while 1:
[... 157 lines cut ...]
        ox.append(-10.0)
        oy.append(i)
    for i in range(-10, 40):
        ox.append(20.0)
        oy.append(i)
    for i in range(0, 40):
        ox.append(40.0)
        oy.append(60.0 - i)

    if show_animation:  # pragma: no cover
        plt.plot(ox, oy, ".k")
        plt.plot(sx, sy, "og")
        plt.plot(gx, gy, "xb")
        plt.grid(True)
        plt.axis("equal")

    bfs = BreadthFirstSearchPlanner(ox, oy, grid_size, robot_radius)

    if show_animation:  # pragma: no cover
        plt.plot(rx, ry, "-r")
        plt.pause(0.01)
        plt.show()

if __name__ == '__main__':
    main()

yashvarshney003, 2020-08-09:
Same error in depth_search.py

UncleBob2, 2020-08-09:
I notice several of them with similar issues. They would create the error at the end of the search when connecting the start to end points.  

Kind regards,
Winsor

yashvarshney003, 2020-08-10:
@AtsushiSakai  Please review the merge request.

Kind regards 
Yash

closed 2020-08-15 by AtsushiSakai