AtsushiSakai/PythonRobotics #365: Hello Sakai-san, firstly, thank you for creating this wonderful repository. Found it to be really helpful.

raghuramshankar, 2020-07-25:
**Describe the bug**
A clear and concise description of what the bug is.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Python version (This repo only supports Python 3.7.x or higher).
 - Each library version

raghuramshankar, 2020-07-25:
Hello Sakai-san, firstly, thank you for creating this wonderful repository. Found it to be really helpful. 
I was wondering if you have any plans on implementing Cubature Kalman Filter for Localization here? 
Is it possible to create a pull request for contributing an entirely new code file, for example, a Cubature Kalman Filter here?

AtsushiSakai, 2020-08-02:
Hi. thank you for your suggestion. Any kind of contributions are welcome. 
But, before that, I want to check some things, because I'm not familiar with the cubature kalman filter.
- What are the pros and cons of the  cubature KF compared to EKF?
- Is this widely used in academic or industrial fields?

raghuramshankar, 2020-08-03:
Hi,

raghuramshankar, 2020-08-03:
Hi,
Thank you for your reply. 
The Cubature Kalman Filter is a relatively new type of Kalman Filter, which uses the Cubature Rule, a modification/special case of the Unscented Transform. Thus, it works in a way similar to the UKF, can be used for highly nonlinear motion models.
Here, we use only 2*n sigma points, and the the zeroth weight is always taken as zero.
Thus, while using a CKF for localization, there are no extra tuning parameters (except for the process noise) and there is no possibility of getting negative weights.
Performance has been shown to be similar to the UKF in most scenarios, but with the above added advantage.
Paper for reference:
[Cubature Kalman Filters](https://ieeexplore.ieee.org/document/4982682)
This was taught me at my university, and we were told that the industry is moving towards using CKF.
It would be great to contribute to this repository, please let me know if it is ok!

AtsushiSakai, 2020-08-11:
OK. Please create a PR for it. Let's discuss details in the PR.

closed 2020-10-09 by AtsushiSakai