AtsushiSakai/PythonRobotics #373: Jacobian calculation in extended_kalman_filter

bender2po, 2020-08-07:
**Describe the bug**
I think, calculation of jacob_f and jacob_h is not proper in extended_kalman_filter.py

**Expected behavior**
Please see the screen-shot for an element in jacob_f.
Also, why dx/dphi term is zero jacob_h but not in jacob_f.?

[image]

AtsushiSakai, 2020-08-11:
Hi. Thank you for your comment.
> why dx/dphi term is zero jacob_h but not in jacob_f.?
Because, the `x` can be observed directly with GPS in observation process.
Ref: https://github.com/AtsushiSakai/PythonRobotics/blob/master/Localization/extended_kalman_filter/extended_kalman_filter_localization.ipynb

bender2po, 2020-08-13:
@AtsushiSakai Thank you for your response. 
1. In the ref, it is given that JF[1,3] = -vsin(phi)dt, hower, as per my above calculations it's JF[1,3] = v/w cos(phi). In extended_kalman_filter.py, it is mentioned that  x_{t+1} = x_t+v*dt*cos(yaw) => dx/dyaw = -v*dt*sin(yaw). here, x_t, v are treated as constants. It doesn't make much sense for me.
2. If 'x' is constant then dx = 0 => dx/dx is undefined. Hence, B[1,1] should be undefined instead of 0.

AtsushiSakai, 2020-08-15:
> here, x_t, v are treated as constants. It doesn't make much sense for me.

Because, JF[1,3] is the partial derivative of yaw, so x, and v is not variable in the term.

>  If 'x' is constant then dx = 0 => dx/dx is undefined

JH[1,1] is is the partial derivative of x, so x is variable, not constant in the term.

closed 2020-08-15 by bender2po