AtsushiSakai/PythonRobotics #420: Unittests display animation

raghuramshankar, 2020-10-10:
**Describe the bug**
When running unittests, the script `A_Star_searching_from_two_side.py` runs its tests and displays animation as well, which makes things slow. Some other unit tests print out information on the terminal.

**Expected behavior**
I think we should run tests without showing animation, print something in the terminal only if a unit test fails.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Python version 3.8.1
 - Numpy version 1.18.3
 - Matplotlib version 3.2.1
 - Scipy version 1.5.1
 - Pandas version 1.0.5

AtsushiSakai, 2020-10-12:
Hi. Thank you for reporting. Can you create a PR to fix it?

raghuramshankar, 2020-10-12:
Sure, I can do that.

PR #429 (2020-11-01): fix unittest animation bugs

#### Reference issue
fix #420 

#### What does this implement/fix?
fix unittest animation bugs

#### Additional information
closed 2020-11-01 by AtsushiSakai