AtsushiSakai/PythonRobotics #425: State lattice planner confusion regarding xc and yc calcuation

CtfChan, 2020-10-20:
Is anyone able to draw a diagram of what this looks like? Why do we need to add `math.sin(l_heading) * l_center` to xc and ` math.cos(l_heading) * l_center` to yc? Shouldn't d and the heading suffice to find xc, yc?

```python
def calc_lane_states(l_center, l_heading, l_width, v_width, d, nxy):
    """
    calc lane states
    :param l_center: lane lateral position
    :param l_heading:  lane heading
    :param l_width:  lane width
    :param v_width: vehicle width
    :param d: longitudinal position
    :param nxy: sampling number
    :return: state list
    """
    xc = math.cos(l_heading) * d + math.sin(l_heading) * l_center
    yc = math.sin(l_heading) * d + math.cos(l_heading) * l_center
```

AtsushiSakai, 2020-11-01:
Hi. Thank you for reporting. Yes I agree that. Can you create a PR to fix it?

CtfChan, 2020-11-03:
Hi. Sure I'll do it this weekend (y)

PR #495 (2021-04-02): fix state_lattice_planner.py coordinate conversion

#### Reference issue

fix #425

#### What does this implement/fix?

#### Additional information
closed 2021-04-02 by AtsushiSakai