AtsushiSakai/PythonRobotics #499: Proposal : improve testing configuration

sjev, 2021-04-09:
Currently the tests are located in `tests` folder which also contains an `__init__.py` file.
It should not be there as it causes quite some import issues as described [here](https://stackoverflow.com/questions/41748464/pytest-cannot-import-module-while-python-can)
Also a `conftest.py` hack is used to make the tests run.

A better and clener way would be to add the library to the python path by installing it with pip in editable mode 

> pip install -e . 

( to do this a `setup.py`  needs to be added to the root folder.)  I've included one in my recent pull request.

The result would be that the modules can be imported during testing without any hacking, like this:

```python
from PathPlanning.AStar import a_star as m

def test_1():
    m.show_animation = False
    m.main()
```

no conftest is needed

AtsushiSakai, 2021-05-03:
Thank you for your proposal!! I'm sorry for late reply.
> 1. make the PythonRobotics library installable by adding setup.py
I think adding setup.py for this repo, it might fix the dependency hack issue like #486
A PR is wellcome!!

> 2. remove conftest.py workaround
If remove the `conftest.py', how can we run the test of a specific test file?

kajal-puri, 2021-10-18:
Hi,

I'm interested to create a setup.py file for the tests directory (and remove __init__.py). 
Just to clarify the structure of the file, I'm eager to follow the scipy [setup.py](https://github.com/scipy/scipy/blob/master/setup.py), not everything in there but somewhat similar. Do you think it will fulfill all our needs or do you have any other suggestions? Thank you.

closed 2023-06-11 by AtsushiSakai