AtsushiSakai/PythonRobotics #505: "smooth" reeds_shepp_path so it keep path straight but it doesnt go through waypoints

lhoupert, 2021-04-28:
Is there an option to keep the paths generated by  `reeds_shepp_path_planning` straight? 

Right now when I run the function on my data I got the results below:

[image]

However, I would like to get something similar to this:
[image]

I don't mind not going through the waypoints. 

<details>
<summary>The full code I am using can be seen here (click to expand)</summary>
<p>

```python
# To add a new cell, type '# %%'
# To add a new markdown cell, type '# %% [markdown]'
# %%
from IPython import get_ipython

# %% [markdown]
# # Test python robotics examples
# %% [markdown]
# https://github.com/AtsushiSakai/PythonRobotics/blob/master/PathPlanning/ReedsSheppPath/reeds_shepp_path_planning.py

# %%
"""
Reeds Shepp path planner sample code
author Atsushi Sakai(@Atsushi_twi)
"""
import math

import matplotlib.pyplot as plt
import numpy as np

show_animation = True

class Path:
    def __init__(self):
        self.lengths = []
        self.ctypes = []
        self.L = 0.0
        self.x = []
        self.y = []
        self.yaw = []
        self.directions = []

def plot_arrow(x, y, yaw, length=1.0, width=0.5, fc="r", ec="k"):
    """
    Plot arrow
    """

    if not isinstance(x, float):
        for (ix, iy, iyaw) in zip(x, y, yaw):
            plot_arrow(ix, iy, iyaw)
    else:
        plt.arrow(
            x,
            y,
            length * math.cos(yaw),
            length * math.sin(yaw),
            fc=fc,
            ec=ec,
            head_width=width,
            head_length=width,
        )
        plt.plot(x, y)

def mod2pi(x):
[... 511 lines cut ...]
    
    # Store results in a dictionnary
    WP_trajectories[k]['xypath']= [[round(x,2),round(y,2)] for x,y in zip(px, py)]

mode

# %%
# unpack dictionary for plotting purposes
trajectories = [list(zip(*traj['xypath'])) for k,traj in WP_trajectories.items()]
trajx = [item for traj in trajectories for item in traj[0]]
trajy = [item for traj in trajectories for item in traj[1]]

# %%
fig = plt.figure(figsize=[12,8])
ax = fig.add_subplot(111)
plt.plot(WPsx, WPsy, 'r+', markersize=12)
for name, xy in zip(WPsname, zip(WPsx, WPsy)):                                     
    ax.annotate(name, xy=(xy[0], xy[1]+0.5) , textcoords='data', va='bottom', ha='center', color='r', fontsize=12)
    
plt.plot(trajx, trajy, 'b.-', markersize=8)

</p>
</details>  

```

AtsushiSakai, 2021-04-29:
No. Reeds shepp cannot be used for it.

closed 2021-04-29 by AtsushiSakai

lhoupert, 2021-04-29:
Thanks. Do you have suggestions about something else I could use?

AtsushiSakai, 2021-04-29:
Spline based approach might help you: https://github.com/AtsushiSakai/PythonRobotics/blob/master/PathPlanning/BSplinePath/Figure_1.png