AtsushiSakai/PythonRobotics #514: Voronoi Roadmap

FCostaS, 2021-06-08:
**Expected behavior**
Could someone explain to me why this occurs?

**Screenshots**
[image]

AtsushiSakai, 2021-06-09:
Thank you for reporting it. That might be a bug. Could you give us the code reproducing the big?

FCostaS, 2021-06-09:
> Thank you for reporting it. That might be a bug. Could you give us the code reproducing the big?
It's here.
[main.zip](https://github.com/AtsushiSakai/PythonRobotics/files/6627306/main.zip)

mbosetti, 2021-06-29:
I ran in to the same issue with the road mapper. It turned out to be a simple hypot check. I assume it will work for this as well

https://github.com/AtsushiSakai/PythonRobotics/pull/522

PR #522 (2021-06-29): fix dijkstra hypot check bug

#### Reference issue

fix #514

#### What does this implement/fix?

Fixes the dijkstra_search where the hypotenuse check was incorrect and returning a false 'same node'

#### Additional information
closed 2021-06-29 by AtsushiSakai