AtsushiSakai/PythonRobotics #515: 'No module named' error for many examples.

benthebear93, 2021-06-10:
`  File "grid_based_sweep_coverage_path_planner.py", line 20, in <module>
    from Mapping.grid_map_lib.grid_map_lib import GridMap
ModuleNotFoundError: No module named 'Mapping' 
`

I was trying to run a grid_based_sweep_coverage_path_planning and encountered ModuleNotFoundError on custom library(Module) that you created. I noticed that you recently made commit(#456) on few examples that related to sys.path.append.
I could run without error after adding sys.path.append to main code(grid_based_sweep_coverage_path_planning.py) from __init__.py

`import math
from enum import IntEnum

import numpy as np
from scipy.spatial.transform import Rotation as Rot
import matplotlib.pyplot as plt
import os
import sys

GRID_MAP_LIB = os.path.dirname(os.path.abspath(__file__)) + \
               "/../../Mapping/"

sys.path.append(GRID_MAP_LIB)

try :
    from grid_map_lib.grid_map_lib import GridMap
except ImportError:
    raise`

I also notice that some of the examples still doesn't have __init__.py file. 
Is there any reason you separate sys.path.append part to __init__.py?
and i don't know if the code is wrong or i am doing something wrong. 

frenet_optimal_trajectory.py has sys.path.append inside its own code works fine.
grid_based_sweep_coverage_path_planner.py, closed_loop_rrt_star_car.py has sys.path.append inside __init__.py code which
gives ModuleNotFoundError.  

**Desktop (please complete the following information):**
 - Python version : 3.8.10
 - Each library version

AtsushiSakai, 2021-06-10:
Hi. These should be fixed. If you are interested in, your PR is welcome.

benthebear93, 2021-06-14:
> Hi. These should be fixed. If you are interested in, your PR is welcome.

I will try to do it in this week or so :)

PR #516 (2021-06-21): Fix No module error in GridBasedSweepCPP and ClosedLoopRRTStart
#### Reference issue
fix #515

#### What does this implement/fix?
changed import Module path.
all the other codes has sys.path.append code in the main python code except
GridBasedSweepCPP and ClosedLoopRRTStart has in __init__.py which makes "No module error"

#### Additional information
closed 2021-06-21 by AtsushiSakai