AtsushiSakai/PythonRobotics #530: Add mission planning examples.

AtsushiSakai, 2021-07-17:
Mission planning is important for complicated robotics tasks. So It would be great to add some simple mission planning examples. For example, 
- Finite state machine
- BehaviorTree
etc.

keep9oing, 2022-01-14:
Task allocation/assignment/planning/scheduling seems to be needed.

AtsushiSakai, 2024-10-13:
PR is welcome, no need to assignment

Aglargil, 2025-02-20:
@AtsushiSakai  Hi, I'm really interested in this task. I plan to provide a simple example where a state machine is used at the top level to manage different modules like **localization**, **mapping**, and **planning**. Inside the **planning** module, a behavior tree will be used to manage the robot's behavior. 

The problem is that the current project doesn't rely on the corresponding state machine and behavior tree libraries. Are you planning to import these libraries? If so, do you have any recommended ones? If you prefer not to add external dependencies, we could also try creating a simple state machine and behavior tree library. What do you think?

AtsushiSakai, 2025-02-20:
Thank you for your interest.

I totally prefer this:
> If you prefer not to add external dependencies, we could also try creating a simple state machine and behavior tree library. 
because this project should focus on education than practicality.

Aglargil, 2025-02-20:
> Thank you for your interest.谢谢您的兴趣。
> 
> I totally prefer this:  我完全更喜欢这个：
> 
> > If you prefer not to add external dependencies, we could also try creating a simple state machine and behavior tree library.如果您不想添加外部依赖关系，我们还可以尝试创建一个简单的状态机和行为树库。
> 
> because this project should focus on education than practicality.因为这个项目应该专注于教育而不是实用性。

Okay, I'll consider how to achieve it in a simple and effective way

closed 2025-05-02 by AtsushiSakai